microsoft/pxt-adafruit #224: Switch move Left fires

Jaqster, 2017-07-07:
I noticed that on startup, the 'Switch Moved Left' event fires (even if the Switch is in the Right side position).
Repro - move the switch to the right, plug in CPX, download something like this...  You'll see the lights go White, Yellow, then Purple.
input.onSwitchMoved(SwitchDirection.Left, () => {
    light.showRing(
    `yellow yellow yellow yellow yellow yellow yellow yellow yellow yellow`
    )
})
input.onSwitchMoved(SwitchDirection.Right, () => {
    light.showRing(
    `purple purple purple purple purple purple purple purple purple purple`
    )
})
light.showRing(
`white white white white white white white white white white`
)

abchatra, 2017-10-25:
@microsoftsam what is the status of this bug. We need this to ship.

microsoftsam, 2017-11-02:
This might be a codal issue, or maybe because we basically re-use the button logic. 
Essentially as the program starts, it needs to keep a note of the current state and only trigger an event if that state is changed..

pelikhan, 2017-11-02:
I kind of remember that we added the logic to trigger that event when registered.

microsoftsam, 2017-11-03:
Yea I chased up the old issue and commented on a possible fix there: https://github.com/Microsoft/pxt-adafruit/issues/128

closed 2017-11-03 by microsoftsam