microsoft/pxt-adafruit #226: Regression in showRing compilation

microsoftsam, 2017-07-11:
There's been a regression in the showRing compilation.
1. Drag a show ring block inside an onstart or forever loop
2. Switch to Javascript. 

This is what the generated JS looks like:
```
    light.showRing(
    red red red red red red red red red red
    )
```

Generated JS doesn't include quotes.

Doesn't repro if you make a change to the ring, like update one of the colors. 

Repro's on /beta

microsoftsam, 2017-07-11:
Related to this change: https://github.com/Microsoft/pxt-adafruit/commit/c17203b140f5d1f639e671ef17bd9344b0dc05ba

guillaumejenkins, 2017-07-12:
The fix created a new regression:

If you have a showRing block with non-default values (i.e. not all red LEDs), doing a round trip causes a bad compilation:

[image]

microsoftsam, 2017-07-12:
Okay, mind looking into this @guillaumejenkins?

guillaumejenkins, 2017-07-12:
Yep, on it!

guillaumejenkins, 2017-07-12:
Fixed in bb7dcc9fc39e34c2f1a2348ff81841111f2cf7e2

commit 48d03782 (2017-07-11): Fix showRing compilation regression. Fixes #226
closed 2017-07-12 by guillaumejenkins