microsoft/pxt-adafruit #229: serial streaming stops working (MKCD app)

tballmsft, 2017-07-11:
Repro:

1. Download this code to connected CPLAY:
```
loops.forever(function () {
    serial.writeNumber(input.acceleration(Dimension.X))
    serial.writeLine("")
})
```
2. Observe serial working
3. Reset the application (via gear menu).
4. Note that serial is no longer working, even though CPLAY continues to stream data.
5. Requires app reboot.

riknoll, 2017-07-13:
Also if you hit the reset button on the circuit playground twice (once for bootloader, once to get back to the program) serial stops working and you need to reboot the app.

guillaumejenkins, 2017-07-14:
Not app specific; this repros in the browser when doing local serve

abchatra, 2017-10-25:
@caitlin-hennessy what is the status of this bug?

caitlin-hennessy, 2017-10-26:
On the app side, this is fixed by the new serial reconnection logic.

However, on the winrt side, there seems to be a sporadic issue with HidDevice.fromIdAsync() - sometimes it returns null even when the Adafruit is plugged in.  We're continuing to troubleshoot this.

caitlin-hennessy, 2017-10-26:
Should be fixed by https://github.com/Microsoft/pxt/pull/3339

abchatra, 2017-11-02:
Fixed

closed 2017-11-02 by abchatra