microsoft/pxt-adafruit #230: some testing todos (MKCD app)

tballmsft, 2017-07-11:
-[ ] Flashing (bootloader mode, non-bootloader mode, unplug / replug device during a session and make sure you can still flash, etc)
-[ ] Writing serial data from the device to the editor (try unplug / replug like above)
-[ ] Saving projects to the system (using the little Save icon next to the project name)
-[ ] Double-clicking project files (.uf2, .mkcd, .pxt, .hex) and make sure the app handles them (double click while app running vs not running, try clicking files from other targets (should handle it correctly, i.e. create a new project instead), try drag & drop file onto the app, try the import button + file picker)
-[ ] Offline (try setting wifi to airplane mode and launch the app, make sure you can do all of the above)

abchatra, 2017-07-11:
Thanks Tom! @guillaumejenkins FYI

tballmsft, 2017-07-13:
Actually, this list was contributed by @guillaumejenkins

abchatra, 2017-07-13:
Done

closed 2017-07-13 by abchatra