microsoft/pxt-adafruit #231: streaming temperature yields bogus values

tballmsft, 2017-07-11:
black time | black data | log time
-- | -- | --
0 | -2147483648 | 0
0.024 | -2147483648
0.048 | -2147483648
0.073 | -2147483648
0.103 | -2147483648
0.121 | 2147483647
0.161 | 2147483647
0.171 | 2147483647
0.201 | 2147483647
0.219 | 2147483647
0.257 | 2147483647
0.284 | 2147483647
0.302 | 2147483647
0.315 | 2147483647
0.337 | 2147483647
0.379 | 2147483647
0.39 | 2147483647
0.41 | 2147483647
0.434 | 2147483647
0.456 | 2147483647
0.48 | 2147483647
0.504 | 2147483647
0.528 | 2147483647
0.552 | 2147483647
0.578 | 2147483647
0.6 | 2147483647
0.624 | -2147483648
0.648 | -2147483648
0.672 | -2147483648
0.699 | -2147483648
0.721 | -2147483648
0.751 | -2147483648
0.769 | -2147483648
0.804 | -2147483648
0.818 | -2147483648
0.85 | -2147483648
0.867 | -2147483648
0.889 | -2147483648
0.915 | -2147483648
0.945 | -2147483648
0.966 | -2147483648
0.985 | -2147483648
1.022 | -2147483648
1.035 | -2147483648
1.057 | -2147483648
1.088 | -2147483648
1.111 | -2147483648
1.138 | 2147483647
1.158 | 2147483647
1.178 | 2147483647
1.201 | 2147483647
1.229 | 2147483647

abchatra, 2017-07-13:
@tballmsft hardware issue?

abchatra, 2017-07-13:
May be we should try with circuit python and see if the issue reproduces there as well.

guillaumejenkins, 2017-07-14:
This repros in the website when doing local serve, so it's not app related. Could it be a codal bug?

mmoskal, 2017-07-14:
Can you attach repro script?

guillaumejenkins, 2017-07-14:
https://makecode.com/_MhDXFp2tUbdM

You'll need to repro it in local serve since serial doesn't work from device to the website

pelikhan, 2017-10-20:
Dup of https://github.com/Microsoft/pxt-adafruit/issues/320

closed 2017-10-20 by pelikhan