microsoft/pxt-adafruit #232: Flashing doesn't work in Bootloader Mode

Jaqster, 2017-07-13:
This is really confusing for users - many I'm sure will be using both the App and the Browser version of MakeCode.  And it's the exact opposite what we're telling them to do:
Browser - Press Reset to go into Bootloader, then Flash
App - Do not press Reset, do not to go into Bootloader to Flash

Flashing through the app should be super simple and easy - plug the device in and download, period.  And actually, should we change the Button text to something else?  Not sure "Download" is the right verb for flashing via an App - "Deploy"?

guillaumejenkins, 2017-07-13:
Weird, flashing works both in bootloader and user space for me... Is it happening to others? Does it still happen after you reboot the app? Does it still happen if you unplug and replug the device?

riknoll, 2017-07-13:
Doesn't work for me either

riknoll, 2017-07-13:
Bootloader version 1.22.0

guillaumejenkins, 2017-07-13:
Ah that explains it; this only works for bootloader 1.23.0, the one that is shipped

riknoll, 2017-07-13:
Looks like 1.22.0 is what is on the ones we got

guillaumejenkins, 2017-07-13:
@ladyada Can you confirm what bootloader version is shipped with the devices?

guillaumejenkins, 2017-07-17:
We'll be supporting both. PR in progress

closed 2017-07-18 by guillaumejenkins