microsoft/pxt-adafruit #234: Examples & Blocks leads to Oops page

abchatra, 2017-07-13:
[image]

Please check all other menu items as well

Jaqster, 2017-07-13:
Also the Code section of the Lucky Wheel and Hot Potato projects is missing too.

guillaumejenkins, 2017-07-14:
This happens in https://makecode.adafruit.com/beta too and is simply because we haven't done a /beta -> live release since adding those in. This will fix itself once we do a /beta -> live release and we make the app point to the live website

abchatra, 2017-07-18:
This is broken in release now. This is due to change of API from random to randomRange and any documentation dependent on that is broken. I am going to do a point fix now to see if this can be fixed.

abchatra, 2017-07-18:
fixed

closed 2017-07-18 by abchatra