microsoft/pxt-adafruit #240: offline documentation error needs improvement

abchatra, 2017-07-13:
[image]

abchatra, 2017-07-13:
[image]

Also share button in offline doesn't give any error message

abchatra, 2017-07-13:
Duplicate of https://github.com/Microsoft/pxt-adafruit/issues/243

guillaumejenkins, 2017-07-14:
What would you like to see as the error page? Something like:

"You appear offline - viewing this page requires Internet access"

Please keep in mind that the "offline page" is the same for the app itself - it basically gets shown by the UWP platform anytime there's no internet access. This could also include the simulator iframe. So, it needs to be generic.

guillaumejenkins, 2017-07-14:
Let's keep this issue about the error page, and #243 for the share dialog

abchatra, 2017-07-14:
Make it something kid friendly. For inspiration https://www.adafruit.com/sfsdf

pelikhan, 2017-07-14:
"Oops, please connect to internet"

guillaumejenkins, 2017-07-14:
[image]

How about this? I think it's useful to include the URL and HTTPS status to help users submit bug reports

abchatra, 2017-07-14:
Looks good. Can you some how blend the background color with the theme of the website?@microsoftsam might have opinions here.

guillaumejenkins, 2017-07-14:
Sure, we can make it a bit more Adafruit-like, but keep in mind it's 100% static - so it can't dynamically fetch theme info or images or etc

guillaumejenkins, 2017-07-14:
[image]

abchatra, 2017-07-14:
:shipit:

guillaumejenkins, 2017-07-14:
Merged to master, but this will require an update to the store, as this is in the app shell

closed 2017-07-14 by guillaumejenkins