microsoft/pxt-adafruit #242: Projects - Make and Examples hang in offline mode

Jaqster, 2017-07-13:
The Make and Examples screen also spin forever in Offline mode.

[image]

guillaumejenkins, 2017-07-14:
Tutorials and examples are considered "documentation", and so are not bundled with the website. They're downloaded on the fly. Same for Getting Started.

I'm not sure how much effort it would take to fix this, but this is in the same boat as all documentation. We should discuss what we want to do about this.

guillaumejenkins, 2017-07-14:
Are we ok with an error message in the dialog until we know how critical offline experience is?

pelikhan, 2017-07-14:
Yes. Projects are full of images and videos. Unlikely we can fully locally cache those.

abchatra, 2017-07-17:
Error is ok for now.

closed 2017-07-18 by guillaumejenkins