microsoft/pxt-adafruit #244: Save button also activates download button

abchatra, 2017-07-13:
When you click save there is a doughnut on the download button as well and it is confusing as it is not flashing here.

abchatra, 2017-07-13:
Download and save were same in website so this is not a severe issue there

guillaumejenkins, 2017-07-14:
This is by design; it's not "activating" the download button, it's merely blocking the button so you cannot hit download while saving. This is because both operations require a compile to happen, so we don't want you to download while saving.

I think it should stay that way, what do you think?

abchatra, 2017-07-14:
Can we disable it instead of doughnut during the save? Otherwise it looks like it is flashing the device.

guillaumejenkins, 2017-07-14:
New experience (in PR) is much better:

[image]

abchatra, 2017-07-14:
Awesome :-)

closed 2017-07-14 by guillaumejenkins