microsoft/pxt-adafruit #248: Compilation issue under Safari in Prod and Beta.

veler, 2017-07-18:
**Expected :** 
On MakeCode for Adafruit on **_Safari under MacOS_**, clicking on Save or Download button must compile the code correctly.

**Current situation :** 
Clicking on Save or Download button throw a "Compilation failed, please check your code for errors".

**Reproduce :** 
- Open [https://makecode.adafruit.com/](https://makecode.adafruit.com/) or [https://makecode.adafruit.com/beta](https://makecode.adafruit.com/beta) on _**Safari for macOS**._
- Create a new project.
- Give it a name.
- Click on Save or Download button.

Same situation if we are in the JavaScript editor.

riknoll, 2017-07-22:
The issue is an old WebKit bug where the function `btoa()` is not defined for webworkers. This doesn't affect newer versions of Safari, but we should probably add a polyfill anyways.

riknoll, 2017-07-22:
Also, this has been around for ~8 months or so.

pelikhan, 2017-07-22:
We should have a polyfill... maybe for another of those conversion functions.

abchatra, 2017-07-26:
@riknoll can you please prioritize this, please?

pelikhan, 2017-11-02:
Fixed.

closed 2017-11-02 by pelikhan