microsoft/pxt-adafruit #266: Accelerometer Y-axis values mixed on Simulator vs. Hardware

Jaqster, 2017-07-27:
This bug was originally reported by a customer.  I didn't understand his code or what he was trying to do, but tried this quick test, and it does look like the Y-axis values are switched on the Hardware:

loops.forever(function () {
    if (input.acceleration(Dimension.X) > 800) {
        light.showRing(
        `white white white white white red red red red red`
        )
    } else if (input.acceleration(Dimension.X) < -800) {
        light.showRing(
        `red red red red red white white white white white`
        )
    } else if (input.acceleration(Dimension.Y) > 800) {
        light.showRing(
        `red red red white white white white red red red`
        )
    } else if (input.acceleration(Dimension.Y) < -800) {
        light.showRing(
        `white white red red red red red red white white`
        )
    } else {
        light.showRing(
        `white white white white white white white white white white`
        )
    }
})
light.showRing(
`white white white white white white white white white white`
)

abchatra, 2017-08-03:
@myjeeed did you see any issues with the accelerometer values?

abchatra, 2017-10-16:
@riknoll this is fixed right? Can you resolve this?

closed 2017-10-16 by riknoll