microsoft/pxt-adafruit #282: When clicking comments firstly, it disappear quickly

WendyFeng01, 2017-08-10:
**Repro Steps:**
• Navigate to https://makecode.adafruit.com/beta
• Drag "show animation" block under forever block
• Right click block and choose "Add Comment", put some comments in it
• Close the comments, click the blank area and open the comments again

**Expected result:**
The comments open again and does not disappear
**Actual result:**
The comments disappear quickly firstly
[image]

Helen09, 2018-03-15:
The issue still repro, but not very stable. It repros after clicking the blank area and open the comments again
[image]

WendyFeng01, 2018-04-26:
Closed as not repro.

closed 2018-04-26 by WendyFeng01