microsoft/pxt-adafruit #289: Block "length of text 'abc' " & "join", which under "TEXT", doesn't have any reference document.

Emma-sunny, 2017-08-17:
**Repro step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Click "_ADVANCE_" and choose "_TEXT_"
- Right click "_length of text 'abc'_ " or "_join_", then choose "_help_"

**Expected result:**
Those blocks should have reference documents.

**Actual result:**
It doesn't have any reference document and the page shows:
[image]

Emma-sunny, 2017-08-17:
Under "_Arrays_", there are some blocks don't have reference documents except "_list add value to end_" and "_remove last value from list_".

ganicke, 2017-08-29:
Missing block docs added in https://github.com/Microsoft/pxt/pull/2841.

closed 2017-08-29 by ganicke