microsoft/pxt-adafruit #304: Accelerometer not working?

ladyada, 2017-08-29:
hiya - i never tried the accel before but its not seeming to work

i tried this sketch from a customer
https://forums.adafruit.com/viewtopic.php?f=58&t=122142&p=612590#p612590
they say the Y dim is wrong, but i actually never get anything at all, i added a blinking red LED and it seemed to 'crash'? 
unclear whats up! :)

abchatra, 2017-08-29:
@riknoll take a peek please.

riknoll, 2017-08-29:
@ladyada can you send me the script that crashed? The axis is definitely flipped, but I couldn't reproduce the crash.

ladyada, 2017-08-29:
its the same one in the thread. im on firefox win 7 if that helps :/

abchatra, 2017-08-29:
@ladyada I can't repro the crash in firefox.  I assume browser is crashing. If yes can you share the version of firefox?

We have fixed the other bug from our end.

ladyada, 2017-08-31:
well im also getting some sort of null error but here ya go:

[image]

https://makecode.com/_e8F4CUHDbFXR

pelikhan, 2017-08-31:
Looks like light.statusLED is gone. Might have been removed with the refactorings.

closed 2017-08-31 by riknoll