microsoft/pxt-adafruit #317: Lights are missing on Edge

WendyFeng01, 2017-09-08:
**Repro Steps:**
• Navigate to https://makecode.microbit.org/beta on Edge
• Drag "show string" block under on start
• Observe the simulator
**Notes:** If you can't repro this, please refresh edge browser

**Expect Result:(IE,Chrome,Firefox)**
[image]
**Actual Result:(Edge)**
[image]

abchatra, 2017-09-11:
What version of edge? I can't repro this

WendyFeng01, 2017-09-11:
@abchatra  , Edge version is 16258.1023.170905-1207.

abchatra, 2017-09-11:
This looks like a bug in Edge rather than us. Works fine with the older versions of the edge for us.

abchatra, 2017-10-16:
Resolving and please reopen if you still see this.

closed 2017-10-16 by abchatra