microsoft/pxt-adafruit #319: some named color variables are are not recognized as equivalent in comparison operators

sheamusburns, 2017-09-12:
Here's a program that demonstrates the bug:

https://makecode.com/_8XpUTq2vw6hk

This program works with testColor set to black, but if you change the value to any other color, the program breaks.

sheamusburns, 2017-09-17:
Testing this further -- does anyone know what kind of object `light.pixels.pixelColor(pixelLoc)` returns? I can test it against another instance of itself and get a `true` or `false`, but when I test it against an instance of `light.rgb(r, g, b)` or `light.colors(Colors.Color)` for example, I cant get it to return true.

sheamusburns, 2017-09-23:
[image]

[image]

[image]

pelikhan, 2017-10-17:
This is due to the brightness being applied to the color when stored in the strip (rather than keeping it separate). So by default, the brightness is low, which means the color stored is dimmer and dimmer -- and eventually falls to zero.

sheamusburns, 2017-10-18:
...So it would seem @pelikhan `light.pixels.pixelColor(pixel)` should return the original color with the brightness levels stripped out of the value.

sheamusburns, 2017-10-18:
Understandable. I guess if the intention of the Circuit Playground Express is to create a device and software package that is meant for projects but also for _teaching folks how to code_ (Adafruit markets this as an educational tool), a behavior this counterintuitive is confusing for learners. I know firsthand trying to teach high schoolers who are running into this. They're _setting_ the light with one color, but getting a different color value back when _getting_ the color.   

Even though it's an expected behavior of the original library, this has been throwing my students off (and other users trying to build with it: see: https://github.com/Microsoft/pxt-adafruit/issues/313 ).

microsoftsam, 2017-10-19:
@pelikhan looking into this with delayed brightness: https://github.com/Microsoft/pxt-common-packages/pull/131

pelikhan, 2017-10-20:
This issue should be fixed in https://makecode.adafruit.com/beta . We apply brightness when sending the buffer to the strip so we keep the original color at hand.

sheamusburns, 2017-10-20:
Brilliant![image]

The kids will love that they can now build projects that check a color based on the value of one of the on-board pixels! Thanks @microsoftsam & @pelikhan. 

That said, I noticed in the beta site (which I've not been using actually) that `light.pixels.pixelColor()` is available in typescript mode, but no longer has a block for the on-board lights. Will that be available? 

S

pelikhan, 2017-10-20:
it will be back as  a top level block: light.pixelColor()

Meanwhile it is under the light > External category

Brilliant![bitmoji]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcamo.githubusercontent.com%2Fe3509e512b543e8b172a9d7bee2b5061fa7148c6%2F68747470733a2f2f72656e6465722e6269747374726970732e636f6d2f76322f6370616e656c2f31303135323634382d3334363336333135335f31322d73342d76312e706e673f7472616e73706172656e743d312670616c657474653d312677696474683d323436&data=02%7C01%7Cjhalleux%40microsoft.com%7C601852c6d6e844ab2fe508d517563cac%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636440581508687192&sdata=AwXPqCkbVXfOSWf%2FbEtsvlxwpNLsucJCuElKEL1lRlY%3D&reserved=0>

pelikhan, 2017-10-20:
pixelColor is fixed. Light.pixels tracked in https://github.com/Microsoft/pxt-adafruit/issues/398

closed 2017-10-20 by pelikhan