microsoft/pxt-adafruit #320: Customer issue - does temperature work?

ladyada, 2017-09-12:
see https://forums.adafruit.com/posting.php?mode=reply&f=48&t=121247

can someone verify thermistor is supported? :)

abchatra, 2017-09-15:
[image]
Here is the program from the report.

abchatra, 2017-09-15:
@caitlin-hennessy Can you take a look when you have some spare time?

pelikhan, 2017-10-20:
Joe has a fix in CODAL so we need to pick up the latest bits. Michal is working on a "cocal-locked" system to target specific commits.

mmoskal, 2017-10-21:
Does this build solve the issue? https://makecode.adafruit.com/app/2665d3163fd760e170a4b5eee9f552694921916a-b211c34609

I'm not sure I'm getting the exact reading - it shows 26, but it's maybe 24 here

microsoftsam, 2017-10-26:
Verified that the last build fixes the issue. I got it down to 13 degrees C. 

[image]

closed 2017-10-26 by microsoftsam