microsoft/pxt-adafruit #327: After adding elements to array, failed to put block “create Array with ” into "set item to".

Emma-sunny, 2017-09-14:
**Repro step:**
- Navigate to [https://makecode.adadfruit.com/beta](url)
- Drag “_create array with_” into content panel and use “_+_” to add elements to the array
- Then let the variable “_item_” equals the array

**Expected result:**
After adding elements, block "_create array with_ '+'" should be put “_set item to_” successfully.
[image]

**Actual result:**
After add elements to array, failed to put the block to ‘set item to’. Details as follows:
[image]

jedgarpark, 2017-12-11:
Hi, I've found a workaround for this by setting the array in JavaScript and then converting back to blocks. Any work on a fix?

Emma-sunny, 2018-04-26:
Closed as fixed

closed 2018-04-26 by Emma-sunny