microsoft/pxt-adafruit #328: Event “on switch  moved ‘right’” is triggered by default.

Emma-sunny, 2017-09-14:
**Repro step:**
- Navigate to [https://makecode.adafruit.com/beta ](url)
- Drag block “_on switch moved left_” into content panel and change ‘left’ to ‘_right_’
- Put “_set all pixels to ‘red’_” into the block 
[image]

**Expected result:**
The event should be triggered when switch move to “_right_”.

**Actual result:**
The event is triggered by default.

microsoftsam, 2017-10-19:
Verified this is an issue. Instead of firing only "When the switch is moved" It fires on start depending on the position of the switch. 
This happens on the board as well, not just in the simulator.

abchatra, 2017-10-25:
Duplicate of https://github.com/Microsoft/pxt-adafruit/issues/224