microsoft/pxt-adafruit #334: The example of "serialwrite buffer" out of border

WendyFeng01, 2017-09-15:
# Issue 1
**Repro step:**
Navigate to https://makecode.addafruit.com/beta on Edge & IE
Drag block “serialwrite buffer” to “forever”
Right click and choose "Help"

**Expected result:(Chrome & Firefox)**
[image]
**Actual result:(Edge & IE)**
[image]

# Issue 2
**Repro step:**
Navigate to https://makecode.addafruit.com/beta on Edge & IE
Expand ADVANCED and click ARRAYS
Observe the blocks
**Expected result:(Chrome & Firefox)**
[image]

**Actual result:(Edge & IE)**
[image]

pelikhan, 2017-09-15:
Does this issue still repro when custom fonts are disabled?

Helen09, 2017-09-19:
What's the custom fonts?

pelikhan, 2017-09-19:
Unrelated to fonts. Sam has a fix coming up.

closed 2017-10-11 by WendyFeng01