microsoft/pxt-adafruit #339: When dragging block, it disappeared.

Emma-sunny, 2017-09-21:
**Repro step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Drag block "_show animation rainbow for 500 ms_" to content panel
- Then try to put it into "forever"

**Additional information:**
- Only repro on IE[11.1032.16258.0]
- Under “LIGHT”, it seems blocks have the same issue, such as: "_set brightness_", "_photo set pen color_",...

**Expected result:**
[image]

**Actual result:**
The block disappeared.
[image]

Mofell, 2017-10-11:
This issue still repro on latest IE11: 11.1020.17001.0

abchatra, 2017-10-20:
Moving to blockly @microsoftsam

closed 2017-10-20 by abchatra