microsoft/pxt-adafruit #344: After clicking "Getting Started", the page keep the status of "starting tutorial..."

Emma-sunny, 2017-09-21:
**Repro step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Click "_Getting Started_"

**Expected result:**
After clicking "Getting Started", the page should be load successfully.

**Actual result:**
[image]

willingc, 2017-10-10:
Is this still an issue? Possibly was due to latency.

The tutorial does load using Safari and Chrome browsers after a short delay. Perhaps make the motion of the flower greater as it is currently very subtle and it doesn't appear as if anything is happening.

cc/ @ptorrone

abchatra, 2017-10-16:
This is fixed

closed 2017-10-16 by abchatra

willingc, 2017-10-17:
Thanks @abchatra for the update.