microsoft/pxt-adafruit #376: Latest Firmware download page on adafruit.com?

pelikhan, 2017-10-17:
Is it possible to link from the uber CPX article (https://learn.adafruit.com/adafruit-circuit-playground-express/adafruit2-uf2-bootloader-details) into a downloadable firmware binary download?

ladyada, 2017-10-17:
sure. what link?

pelikhan, 2017-10-19:
We cannot host the compiled bootloader. Maybe you can have it on the CPX guide in the bootloader section?

ladyada, 2017-10-19:
to clarify - you mean not the update.uf2, the .bin for jlink-ing?

pelikhan, 2017-10-19:
The .uf2 file that upgrades the bootloader would be best since updating the bootloader is as easy as a drag and drop.

ladyada, 2017-10-19:
its already there? 
https://learn.adafruit.com/adafruit-circuit-playground-express/adafruit2-uf2-bootloader-details#updating-the-bootloader

pelikhan, 2017-10-19:
Oh snap, sorry for wasting your time. I'll cross link that page from our docs.

pelikhan, 2017-10-19:
@ganicke let's find good spots in the docs to link to https://learn.adafruit.com/adafruit-circuit-playground-express/adafruit2-uf2-bootloader-details#updating-the-bootloader which contains a download to the bootloader.

ladyada, 2017-10-19:
lol its all good - theres *a lot* of docs :)

ganicke, 2017-10-20:
We have _/device/usb_ but there's no **Device** in the summary tree. Add Device page with links to these kinds of things?

ganicke, 2017-10-25:
Links added in #427.

closed 2017-10-25 by ganicke