microsoft/pxt-adafruit #377: program terminates on CPX, not in simulator

tballmsft, 2017-10-17:
https://makecode.com/_M4rekjMRX7eL

mmoskal, 2017-10-18:
Can't repro. Are you running from localhost or from the website?

mmoskal, 2017-10-18:
It's possible (though not very likely) this is the timing bug, since it affects USB. On the device that fails, try running it from a USB battery or charger. @tballmsft

tballmsft, 2017-10-18:
Running from windows app..

pelikhan, 2017-10-18:
If this is. Windows app issue, it is P2.

mmoskal, 2017-10-18:
Please try running from a well charger please (on the same board where it hangs). Also try a different board.

tballmsft, 2017-10-18:
Will do.

tballmsft, 2017-10-18:
It was specific to the board - works with another board. What should I do with the buggy board?

mmoskal, 2017-10-18:
The board requires a patch in the clock init code as discussed with Joe. Once he implements it, it should work. Mark it and keep for testing of the fix.

tballmsft, 2017-10-18:
Got it. Thanks.

finneyj, 2017-10-19:
This patch is now applied to master of ```codal-circuit-playground```

Can we pull this through to a beta site or localhost testing for this board, to confirm this addresses the issue?

tballmsft, 2017-10-19:
I'll test it.

mmoskal, 2017-10-19:
@tballmsft - it's somewhat non-trivial, since we're on an older branch of CPX Codal and also because there's another clock init code in the USB code. I would like to sort out the build system first, before we move forward with codal on CPX.

mmoskal, 2017-10-31:
@tballmsft can you test now?

I still see there's two copies of the clock init code - I thought there was only one left.

mmoskal, 2017-10-31:
In CPX v1.2.6 I've removed duplicate code.

 I've uploaded web app using it at https://makecode.adafruit.com/app/492c798798098523aba829daa5424f812bcf7adc-3f85280ecf

Please test both this and the main site.

pelikhan, 2017-11-02:
@mmoskal is that fix in master already?

pelikhan, 2017-11-06:
@tballmsft could you test again with your board? You can go to https://makecode.adafruit.com/beta to pick up the fix. Closing for now.

closed 2017-11-06 by pelikhan