microsoft/pxt-adafruit #378: Buttons/logos missed on IE

Mofell, 2017-10-18:
**Repro Step:**
•	Navigate to https://makecode.adafruit.com/beta on IE
•	Click “New Project”

**Additional Info:**
This issue only repro on IE11
IE11: 11.1024.17001.0

**Expect Result:**
Buttons/logos should display as Edge, Chrome and Firefox
[image]
[image]

**Actual Result:**
Buttons/logos are missing on IE
[image]
[image]

microsoftsam, 2017-10-19:
I dont' repro.. Please try latest.

Mofell, 2017-10-20:
Can't repro on latest build, so close this bug.

closed 2017-10-20 by Mofell