microsoft/pxt-adafruit #383: Need migration story for non-namespaced photon code

pelikhan, 2017-10-19:
[image]

microsoftsam, 2017-10-19:
This upgrade regex might help:
```

                    "light\\.pixels\\.showAnimation": "light.showAnimation",
                    "light\\.pixels\\.showAnimationFrame": "light.showAnimation",
                    "light\\.pixels\\.photonFlip": "light.photonFlip",
                    "light\\.pixels\\.photonForward": "light.photonFlip",
                    "light\\.pixels\\.setPhotonColor": "light.photonFlip",
                    "light\\.pixels\\.setPhotonMode": "light.photonFlip",
                    "light\\.pixels\\.setBrightness": "light.setBrightness",
                    "light\\.pixels\\.graph": "light.graph",
                    "light\\.pixels\\.setPixelColor": "light.setPixelColor",
                    "light\\.pixels\\.setAll": "light.setAll"
```

microsoftsam, 2017-10-19:
https://github.com/Microsoft/pxt-adafruit/pull/391 and https://github.com/Microsoft/pxt-common-packages/pull/133 should fix both issues.

microsoftsam, 2017-10-20:
We also need to add these to the upgrade APIs as it's possible to have old blocks, switch to JS, and not update.

```
light.pixels.__showAnimation(light.animation(LightAnimation.Rainbow), 500)
light.pixels.__setAll(Colors.Blue)
light.pixels.__stopAllAnimations()
light.pixels.__photonForward(1)
light.pixels.__photonFlip()
light.pixels.__setPixelColor(0, Colors.Red)
light.pixels.__graph(0, 0)
light.pixels.__move(LightMove.Rotate, 1)
light.pixels.__setPhotonMode(PhotonMode.PenUp)
light.pixels.__setBrightness(20)
```

pelikhan, 2017-10-20:
Now we need a test plan :)

closed 2017-10-20 by pelikhan