microsoft/pxt-adafruit #390: Monaco toolbox shows two functions for the same method

microsoftsam, 2017-10-19:
This is due to two issues: 
- Firstly, subcategories are not supported in the toolbox
- Secondly, the OO stuff is assuming that the API uses defaultInstance and needs to be fixed to support something like a placeholder field and append the function to it in its snippet. eg: 
dragging out a graph block from the external category should insert something like: strip.graph(0,0) with strip highlighted 

[image]

pelikhan, 2017-11-03:
@abchatra  I don't think this is blocking the release.

microsoftsam, 2017-11-06:
Fixed.

closed 2017-11-06 by microsoftsam