microsoft/pxt-adafruit #392: Cable received api are confusing

abchatra, 2017-10-19:
[image]

As it is very hard to set them up, can we move them to its own package? And keep just the infrared here.
Also remove the cog where here?

abchatra, 2017-10-19:
And the help for these talk nothing about how to set them up. They talk about infrared.

pelikhan, 2017-10-20:
Removed.

pelikhan, 2017-10-23:
Cog wheel still there. Cable removed

pelikhan, 2017-10-25:
https://github.com/Microsoft/pxt-common-packages/pull/144

pelikhan, 2017-10-25:
There are 2 bugs that would require PXT support, probably from @riknoll 

- [ ] unable to specify that at least one argument (num) should be expanded in the block. I added ``?`` to the 2 other arguments to hint that those are optional. Maybe we can use that info.
- [ ] deleting any variable from the signature deletes the entire event.

riknoll, 2017-10-25:
For the first point, there is a way to do that! Add `toolboxVariableArgs="1"`. If you want to have more copies of the block in the toolbox, separate them with semicolons (`toolboxVariableArgs="0;1;2"`).

For the second one, yeah I need to fix that.

abchatra, 2017-10-25:
I do think @microsoftsam and I discussed this and we should give one option to receive number as there no way to send two numbers using the send api.

pelikhan, 2017-10-25:
Right.... There is a typescript API to send an array of numbers but it's not surfaced in blocks.

abchatra, 2017-10-25:
Makes sense. Lets just remove the gear wheel and keep one option to receive number.

pelikhan, 2017-10-25:
There is no gearwheel anymore.

closed 2017-10-26 by pelikhan