microsoft/pxt-adafruit #400: Repetition of the category name at the top

abchatra, 2017-10-20:
[image]

What is the reason to repeat the category name again at the top of the flyout? Can we claim that real estate?

microsoftsam, 2017-10-20:
It's more obvious which category is selected (for the desktop user) that way, and very valuable in mobile views when text is tiny in the category or non-existent.

abchatra, 2017-10-20:
Can we show only in those views then?

microsoftsam, 2017-10-20:
Nope.
The flyout logic is decided in the JS. so we have two choices: 
- only show in real mobile devices (detect that's a mobile device)
- show in all scenarios

The issue with going with the first option is that it's possible to be in the mobile view, and not on a mobile device (ie: resizing your window)

abchatra, 2017-10-20:
Our primary view is 1024x view. I am fine with the first option. I will keep the bug open. P2 though.

microsoftsam, 2017-10-20:
Before we make the call to revert, the original case was on mobile devices only, then we decided to switch to this mode. so lets dig up the reasoning and make a call together. I’d hate to keep going back and forth. There is also a WW dependency here and what they want as this will affect them.

abchatra, 2017-10-20:
Yes I am curious to understand the reasons as well.

abchatra, 2017-10-25:
@microsoftsam Did you have any more information on reasons?

microsoftsam, 2017-10-25:
Unfortunately these discussions were made in person. I'll add a few extra reasons from memory but let's discuss this in person at our design meetings. We should also include the WW designers in the mix.
- Toolboxes are not consistent across targets, with some it's harder to decode the name just due to the text choices and color choices used. Having a heading label at the top makes it clear what category is open, and in a consistent place. 
- Reading the flyout category name is in a consistent place. Unlike the toolbox that is constantly moving based on mouse click. Better readability that way. 
- This way the experience is the same across devices and with plenty more screen realestate on a desktop, the tradeoff for better readability makes sense.
- Finally, it adds structure that kids are used to when reading. Heading on top, and then subheadings (in some cases)

@pelikhan @cwest-ww thoughts? remember any other reasons we decided to keep it on all platforms?

cwest-ww, 2017-10-26:
@microsoftsam Thanks for the heads up on this topic.

This is very important for mobile devices where the css stops showing the category name in the toolbox, otherwise not needed.

I think this is unrelated but I do want to mention that for WW we do need to keep our sub-titles.

pelikhan, 2017-10-26:
I don’t really notice it and ir is very helpful on mobile with collapsed categories buttons.

abchatra, 2017-10-26:
Request was to keep for mobile view and remove it for desktop view where the category name is very clear on the toolbox. Is consistency important here?  It takes up real estate on the low resolution screens.

abchatra, 2017-10-26:
Discussed with @microsoftsam 
Here is the argument to keep it:
"Students blindly click on the categories and scan the blocks. It makes sense to keep the category name at the top on assisting in understanding the  block category"

closed 2017-10-26 by abchatra