microsoft/pxt-adafruit #407: serial.writeNumber doesn't separate numbers in the serial editor

microsoftsam, 2017-10-22:
A simple: 
```
serial.writeNumber(20) 
```

brings up the serial editor but doesn't log numbers on the graph properly. 
It shows 2020202020 as a single value. 

[image]

pelikhan, 2017-10-22:
By design, write number does not add a new line

[screen shot 2017-10-21 at 5 22 47 pm]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F16690124%2F31856935-80f401ce-b684-11e7-812b-d1b309dd40a3.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C5b3194d9ffdf4aa6428608d518e31561%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636442285954735475&sdata=wqNiBGK4Mu750QXKSF80zzGhUjMCsuguhO10YfAQibs%3D&reserved=0>

microsoftsam, 2017-10-22:
Maybe it should show up in the console instead?

caitlin-hennessy, 2017-10-23:
Yes.  I think we should only graph data that comes in as a key:value pair, e.g., "x:123".  Everything else should go to the console.

pelikhan, 2017-10-23:
led.plotBarGraph sends a number per line to the console and automatically gets charted. We should not chart numbers are 256 characters long.

caitlin-hennessy, 2017-10-23:
If we take that approach, we should define a global serial buffer length and use it throughout the code.  Right now the local serve, windows app, and simulator can each have a different buffer length - we can't count on its being 256.

Seems unnecessary to me though.

pelikhan, 2017-10-23:
A sequence of numbers is the simplest way to generate a data stream. We support that kind of output in led.plotBarGraph and light.graph. In this case, we need to detect that the number is obviously too big (e.g. too many significant digits to be representable in JavaScript) and ignore it.

abchatra, 2017-10-25:
@caitlin-hennessy what is the right outcome for this bug?

pelikhan, 2017-11-03:
Follow up bug: https://github.com/Microsoft/pxt-adafruit/issues/469

closed 2017-11-03 by pelikhan