microsoft/pxt-adafruit #416: Playing tones and sound is very limiting. Can we expand this to

microsoftsam, 2017-10-24:
Can we expand this to other sounds like:

microsoftsam, 2017-10-24:
What would it take for us to support samples? @pelikhan @mmoskal @riknoll

pelikhan, 2017-10-24:
The synth infrastructure should be able to play synth. We need to implement the C++ part that takes a sound buffer and streams it. We have something similar with the tone generation where we can take a sample of sound and replay that sound instead of a perfect square wave. So basically, work needed in CODAL.
Once we have the infrastructure, we could have a custom editor that allows to record and encode a sound sample small enough to fit in the flash. We'd probably need support for that 2Mb flash memory on board if we have many of them.

mmoskal, 2017-10-24:
The synth is supported in codal, so the problem would be mostly storing the sounds. Would be good to support the SPI flash for that but for small samples it would be enough to use the built-in one I guess.

microsoftsam, 2017-10-24:
@mmoskal can you point me to what API I could use in codal to achieve this, for now I'm happy with just a sound clip encoded in some string, and we play it. We can wrap blocks around those built ins

pelikhan, 2017-10-24:
music.setTone lets you set the sample used to generate tones but that is not quite what you want.
In any case use Buffer and hex literal for this kind of stuff.

mmoskal, 2017-10-28:
`SAMD21DAC` has `play` method that takes a buffer. There's also `setSampleRate` on it.

pelikhan, 2019-03-13:
We now support mixer in CPX - so multiple cords. No .wav files yet.