microsoft/pxt-adafruit #419: After changing project name, blocks can't be edited on content pane

Helen09, 2017-10-24:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta
• Go to setting and choose "Project Setting"
• Delete project name and click save button
• Type a new name for the project in project setting box and click save button

**Expected result:**
After changing project name, blocks can be edited and blocks show in editor
**Actual result:**
After changing project name, blocks can't be edited on content pane
[image]

microsoftsam, 2017-10-24:
Issue here is that we're allowing you to edit a project name to null.

Helen09, 2017-10-25:
New name was edited and simulator can show, but blocks on content pane still can't be edited

abchatra, 2017-10-25:
Hang seems bad for users.

closed 2017-10-31 by microsoftsam