microsoft/pxt-adafruit #424: Something is leaking memory..

microsoftsam, 2017-10-25:
A simple program like this that doesn't do any fancy animations or anything after a while stops running, and the external Neopixel strip is left with the last remaining strip configuration. 
This behaviour is pretty consistent, I have an inkling that the NeoPixel library might be the culprit here.

```
let pixels = light.createStrip(pins.A1, 100);
loops.forever(function () {
    pixels.setAll(light.colors(Colors.Blue));
    loops.pause(5000);
    pixels.setAll(light.colors(Colors.Red));
    loops.pause(5000);
})
```

microsoftsam, 2017-10-25:
I have tested this on onboard NeoPixels, same behaviour.

pelikhan, 2017-10-26:
I'm running this one for the last 5 minutes with https://github.com/Microsoft/pxt-common-packages/pull/148 and it still running.

```
loops.forever(function () {
    light.setBrightness(20)
    light.setAll(0xFF0000)
    loops.pause(1)
    light.onboardStrip().range(0, 2).setBrightness(0)
    light.setAll(0x33ff33)
    loops.pause(1)
})
```

pelikhan, 2017-10-26:
This one too is happy:
```
loops.forever(function () {
    light.setBrightness(20)
    light.setAll(0xFF0000)
    loops.pause(1)
    light.onboardStrip().range(0, 2).setBrightness(0)
    light.setAll(0x33ff33)
    loops.pause(1)
})
let pixels = light.createStrip(pins.A1, 100);
loops.forever(function () {
    pixels.setAll(light.colors(Colors.Blue));
    loops.pause(1);
    pixels.setAll(light.colors(Colors.Red));
    loops.pause(1);
})
```

pelikhan, 2017-10-26:
It does visibly slow down after a few minutes so something is up.

microsoftsam, 2017-10-26:
I think my test case has been about 10-20 minutes. so it's not a short period of time for you to notice, but I've had a consistent hit where that happens with the code above.

pelikhan, 2017-11-01:
This one freezes after a while.
```
const strip = light.createStrip(pins.A1, 300);
strip.range(0, 100).setAll(Colors.Green);
let brigthness = 64;
let d = 15;
let c = Colors.Green;
strip.setBuffered(true)
loops.forever(function () {
    brigthness += d;
    if (brigthness > 255 || brigthness < 32) {
        d = -d;
        brigthness = Math.max(32, Math.min(255, brigthness));
    }
    strip.setBrightness(brigthness);
    strip.move(LightMove.Rotate, 5)
    strip.show();
    light.photonForward(1)
})
```

pelikhan, 2017-11-01:
I'm looking into this. I see that we're leaking references in the simulator so patching this part.

pelikhan, 2017-11-01:
@mmoskal FYI

pelikhan, 2017-11-01:
https://github.com/Microsoft/pxt/tree/refdebug -- I've moved the ref counting in the runtime as we were leaking between runs.

pelikhan, 2017-11-01:
https://github.com/Microsoft/pxt/pull/3401

mmoskal, 2017-11-02:
It's also possible to track references on the board - there's a #define for that.

pelikhan, 2017-11-03:
Tracked in CODAL.

abchatra, 2017-11-08:
fixed

closed 2017-11-08 by abchatra