microsoft/pxt-adafruit #433: Add light.randomColor

microsoftsam, 2017-10-28:
I think it would be useful to add that to our library of colors.

It would be cool to just have a block that returns a random color.

@pelikhan @Jaqster thoughts?

microsoftsam, 2017-10-28:
Right now I'm thinking something that wraps this: 
```
function randomColor() {
    let rand = Math.randomRange(0, 255);
    return light.hsv(rand, 255, 255);
}
```

pelikhan, 2017-10-28:
Sounds good to me. Maybe we should limit to a palette of clearly différentiable colors.

microsoftsam, 2017-10-28:
Yeah. If only there was a way to determine that programatically..

Jaqster, 2017-10-29:
I like it.  Would be even cooler if we had a generic Random block/function built off our Random Math block that could output random light colors and random music tones as well...