microsoft/pxt-adafruit #442: Download instructions

microsoftsam, 2017-11-01:
We've been struggling with this one a lot, and have found that to be a recurring problem in the classroom: Kids knowing how to get their code onto the device. 
The current experience that just throws text on the screen hasn't seemed to work very well. And you're found having to re-iterate what they need to do to get code on there. 

Kids generally don't read instructions, so using illustrations / images / gifs would best way to get something across (although we should have text there as a backup for those that don't quite get it)

Here's an example of how illustrations work well at conveying the instructions needed: (Chibi Chip upload instructions)
[image]

The steps we feel that would work best for the CPX are: 
- Plug the cable in
- Hit reset to go into "Programming mode", all green lights
- Transfer your downloaded file onto the CPLAYBOOT disk (drag and drop)

Would it be possible to have the Adafruit designers work on some illustrations that can be used for this three step download.

abchatra, 2017-11-02:
This issue (Download instructions) are higher priority than home screen (https://github.com/Microsoft/pxt-adafruit/issues/441)

pelikhan, 2017-11-14:
The latest images can be tested on the beta (https://makecode.adafruit.com/beta). My feedback is that I can't see the purple lines over the green background.
[image]