microsoft/pxt-adafruit #469: Rename Serial to Console.

abchatra, 2017-11-03:
Category : Console
Serial write line:  log
Serial write value:   log value

Change 
"Show data" to "Show console"

pelikhan, 2017-11-03:
We should integrate those APIs to the micro:bit as well to stay consistent.

abchatra, 2017-11-03:
Microbit we can't change the API for v0.

pelikhan, 2017-11-03:
We can add APIs.

abchatra, 2017-11-08:
@caitlin-hennessy 

Still couple of issues pending

1. Monaco toolbox still shows serial but with no blocks.
[image]

2. Show data is not changed to show console.

3. Can console.log can take numbers as well?

pelikhan, 2017-11-08:
(3) This would require special compiler support. We don't support any or unions in the block signature.

abchatra, 2017-11-08:
Lets move that to a separate issue.

pelikhan, 2017-11-08:
@microsoftsam does the Monaco toolbox handle empty namespaces?

pelikhan, 2017-11-08:
https://github.com/Microsoft/pxt/pull/3438

microsoftsam, 2017-11-08:
If it doesn't it should. It should pretty much match the behaviour of the Blocks toolbox. If at any point that's not the case, it's a bug.

pelikhan, 2017-11-08:
It's a namespace where all the blocks are "blocksHidden". This needs to be fixed in the pxt-monaco I guess.

pelikhan, 2017-11-08:
@abchatra I don't think this is a ship blocker.

microsoftsam, 2017-11-08:
The monaco toolbox lives in PXT. so should be an easy fix.

pelikhan, 2017-11-08:
@microsoftsam toolbox issue is yours. I'll look into those missing project pages

microsoftsam, 2017-11-08:
Sounds good

pelikhan, 2017-11-08:
Fix coming in next bump.

closed 2017-11-08 by pelikhan