microsoft/pxt-adafruit #472: IR seems to be broken with latest version

riknoll, 2017-11-03:
Wrote a simple program that sends an index to another CP in a forever loop (once a second) and had the other plot the received number. Barely any of the packets made it through.

mmoskal, 2017-11-03:
Does the same program works with *the same devices* with the previous version?

mmoskal, 2017-11-03:
Also try running both from battery as usb clock will alter behavior now

riknoll, 2017-11-03:
Yep, program works fine with non-beta editor. I also tested on battery and got the same results.

riknoll, 2017-11-03:
Program for reference: https://makecode.com/_hqTC9yChRU5M

mmoskal, 2017-11-06:
I did some digging and it seems with the new clock init, the timers on battery are awfully wrong (10% variance).

I have new message encoding code, will PR soon.

microsoftsam, 2017-11-08:
@mmoskal what's the status of this. I tried with v1.2.8 and it's still very temperamental.

mmoskal, 2017-11-08:
There is a branch in common packages, basically ready to go, might just need disabling debug. Still it has issues locking up the board after a while while sending , but so does the current one.

mmoskal, 2017-11-08:
Should be fixed in https://github.com/Microsoft/pxt-common-packages/pull/169

pelikhan, 2017-11-08:
Tested as fixed on hardware (USB & battery)

closed 2017-11-08 by pelikhan