microsoft/pxt-adafruit #477: Bring back the Show Animation for duration block

Jaqster, 2017-11-03:
Hot Potato project requires this.  We can push it down to the More subcategory, but this project is already out and Boys and Girls Clubs are using it.

pelikhan, 2017-11-03:
Can you attach a shared script url of that project?

microsoftsam, 2017-11-03:
I agree with @Jaqster. There's value in being able to control how long an animation is played for. I don't think it needs to be at the bottom of the list, but certainly warrants another block right next to show Animation frame that has a duration field with it. 

https://makecode.com/_9YHF3ydqib6b

microsoftsam, 2017-11-03:
and it also makes the upgrade experience better, the previous showAnimation block can be converted into that block, and all new blocks that only show an animation with a duration can use the new block.

pelikhan, 2017-11-03:
Let’s bring back the duration in blocks then.

pelikhan, 2017-11-03:
@Jaqster if we bring back the duration, is that enough?

pelikhan, 2017-11-03:
How about this? 
[image]

pelikhan, 2017-11-03:
See https://github.com/Microsoft/pxt-adafruit/tree/showanimationduration

pelikhan, 2017-11-03:
I think we could have a special time picker for animations with better values. In particular, 0 == automatic.

microsoftsam, 2017-11-03:
I think we should split into two blocks. 
There's value in having a simple show animation block with no duration

pelikhan, 2017-11-03:
In this case, the hot potato tutorial is using the instance version -- not the namespace level version. I'm not even sure this fixes the issue.

pelikhan, 2017-11-03:
We could be smart about continuing the previous animation instead of restarting.

microsoftsam, 2017-11-04:
That would be auto upgraded to light.showAnimation, so not really an issue.

pelikhan, 2017-11-04:
I’d like to keep a single “show Anomatikn” block

pelikhan, 2017-11-04:
Having 2 is just going to be confusing.

pelikhan, 2017-11-04:
Now we do have the icon instead of the text....

Jaqster, 2017-11-04:
I like the simplicity of having the Show Animation at the top, like the Show Ring.  Having an extra duration field adds complexity and is one more thing you have to explain and set a value for.

I think it's fine to have multiple blocks doing different things with an animation (frame, duration).

pelikhan, 2017-11-04:
Under the hood, the show animation block still has a duration for app compat. So I don’t think we can have 2 blocks.

pelikhan, 2017-11-06:
https://github.com/Microsoft/pxt-adafruit/pull/480

pelikhan, 2017-11-06:
https://github.com/Microsoft/pxt-common-packages/pull/163

pelikhan, 2017-11-06:
It's back.

closed 2017-11-06 by pelikhan