microsoft/pxt-adafruit #485: There are lot of projects don't have the code instruction.

Emma-sunny, 2017-11-07:
**Repro step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Choose project "_Watch Timer_" under "_Costumes_" and show the projects instructions
- Click "_code_" which displayed on the page's right

**Expected Result:**
It should have the concrete detail after clicking "_code_".

**Actual Result:**
There are lots of project have same issue that the content of code not exist, such as Magic Wand, Reaction!...
[image]

abchatra, 2017-11-08:
I guess this will be in release when we release the docs a s well.

microsoftsam, 2017-11-09:
We have switched to showing instructions in a new tab. This behaviour is now outdated.

closed 2017-11-10 by pelikhan