microsoft/pxt-adafruit #487: Blocks can't be scrolled on IE

WendyFeng01, 2017-11-08:
**Repro steps:**
## Issue 1
• Navigate to https://makecode.adafruit.com/beta
• Click "LOGIC"

**Expected result:**(Edge)
There should have scroll bar
[image]

**Actual result:**(Chrome,Firefox,IE)
There is no scroll bar
[image]

## Issue2
• Navigate to https://makecode.adafruit.com/beta
• Click "LOGIC" and scroll down the blocks

**Expected result:**(Edge,Chrome, Firefox)
The blocks can be scrolled
[image]
**Actual result:**(IE)
The blocks can't be scrolled, and the blocks show imcomplete
[image]

pelikhan, 2017-11-08:
Great catch.

microsoftsam, 2017-11-08:
Fixed in https://github.com/Microsoft/pxt-adafruit/pull/486

pelikhan, 2017-11-08:
Still repro for me on last master.

microsoftsam, 2017-11-08:
Weird, it only repro's for the Logic category.. Interesting

pelikhan, 2017-11-08:
Only one with groups

microsoftsam, 2017-11-08:
Nah, light has groups as well.

microsoftsam, 2017-11-08:
Repro's on minecraft as well, so not really a recent regression. Very intriguing!!

pelikhan, 2017-11-08:
Works in Seeed

microsoftsam, 2017-11-08:
It doesn't work in Seeed for me

pelikhan, 2017-11-08:
right... IE...

microsoftsam, 2017-11-08:
This is not a Z-index issue or anything of that kind. Looks like a Blockly issue as the scrollbar is not even shown in the DOM. Will investigate further tomorrow

microsoftsam, 2017-11-08:
Repro's in the Blockly playground

microsoftsam, 2017-11-08:
Logic, External, Text and Arrays repro. So far I've narrowed down the issue to: 
- If / Else blocks 
- and any block that contains a get_variable block
If the category includes any of the above, it breaks the flyout height logic.

microsoftsam, 2017-11-08:
PXT-blockly issue: https://github.com/Microsoft/pxt-blockly/issues/84

Still investigating, but not only does the toolbox scrollbar break if the blocks listed above are in the Flyout, additionally, when any of those blocks are dragged into the workspace, they disappear as they're dragged. It's all connected, just gotta get to the root cause!

closed 2017-11-09 by microsoftsam