microsoft/pxt-adafruit #488: Conflict with Button A, B, A+B and Pin A1, A2, A3 Click Events

sheamusburns, 2017-11-08:
My students stumbled upon this bug today. 

When you have both and input `On Button A Click` and `On PinA1 Click`, and then you click button A in the sim, it runs the code in the `On PinA1 Click` block. 
`On Button B Click` runs the `On PinA2 Click` block.
`On Buttons A + B Click` runs the `On PinA3 Click' block. 

I have not tested on actual board and have not tested other pin/button combo conflicts. 

Code Sample: https://makecode.com/_i82H7VD90K9m

pelikhan, 2017-11-08:
Thanks for the report! This looks fixed on https://makecode.adafruit.com/beta

sheamusburns, 2017-11-08:
That's awesome. I just verified on the beta. I'll let them know. 

Any idea when the beta if/when beta will replace the current version?

pelikhan, 2017-11-08:
End of week hopefully. We still have a few blocking regressions to address. What’s your feeling on the /beta?

abchatra, 2017-11-08:
@sheamusburns Send us feedback. We love to hear.

closed 2017-11-08 by abchatra

sheamusburns, 2017-11-08:
@pelikhan @abchatra 

I've had the chance to look at the beta a little bit. One thing I love about makeCode is that it really does make it easier to teach kids how to program, and the kids really love it too! They just finished a Mastermind(ish) game challenge where they had to design a guessing game similar to mastermind on the Circuit Playground Express (CPE) and the next challenge they're going to do is to design and build a digital safe, using pins for buttons (maybe conductive paint buttons) and servos.

Most of these kids are brand new to programming, so what I've noticed a couple things about the platform in general. Here are my positives and negatives:

1. Arrays are a hard concept to grasp with the block syntax. Pop and Push methods in JS are much easier to understand in JS. They really have a tough time conceptualizing an array with the language used on the array blocks. I ended up showing a bunch of them JS arrays and they got it better. 

1a. Create Empty Array. I sometimes have kids work in blocks, then go take a look at the TS. When they swap back and forth, sometimes something weird happens with their arrays, like all of the suddenly there will be a new block for an array they created: `create empty array`. Sometimes if the kids swap back and forth between the block and TS modes, repeat `create empty array` blocks will be created for the same array. Kinda bizarre. If the `create empty array` block is necessary for the interpreter and the TS, maybe make that block available?

1b. I love all the new functions available for arrays in the blocks!

2. using the word index in the `for loop` makes kids think it's for arrays because we usually only talk about indexes when referencing items in an array. While you can use the `for index of x to y` block in conjunction with an array, it has more versatility than that and I guess it's just something I noticed about how they were conceptualizing things. 

3. I've noticed in the beta and the stable version that sometimes code just disappears. While it's supposed to save automatically to local storage, there are weird moments where suddenly all the blocks are empty and there's no way to get them back. This has happened with at least three of my students at different times. Most of the time they're using chrome. I've had it happen to me when I was swapping between tabs, but honestly I don't know what behavior is causing this bug.

4. Console! The console is one of the most useful tools in learning to code. As values change during the runtime of a program, it's so useful to be able to print things out. In the production version, the serial tools are limited to writing a line or two but only seeing the full console if you opened up devtools (even then you have to filter out all the other noise in the JS console.). I like that the Serial Log or whatever we're calling it is bigger in the beta. What would be great though is if you could see your code and the Serial Log side by side. A lot of times, I'll point out a bit of code to a kid, print a value to console and then have the kid compare what they're seeing print out in the console to what they think is happening in the code. It's incredibly useful if the Console and the code are visible simultaneously. 

5. NeoPixels. Support for the Neopixel library is awesome. It would be great if we could plug and play in the WS2801 Library too!

6. I love the new if/then block format with the minus sign and the plus sign. So much better than the little gear. Only thing, I'd improve is when you pull an `if true then` block out without an else, then add an else, you can't remove the else and turn it back into just an `if true then` block. That would be nice. 

7. Functions are awesome, yet they would be infinitely better with params :) Right now I can only teach kids that functions are chunks of code that can be reused. It would be cool to be able to pass parameters and also for a function to return something. 

Lol. Ok -- that's more than enough from me. I can't tell you guys how much this platform has changed how I teach coding -- it's fantastic in so many ways and you're doing a great job making it better. Let me know if you ever need more feedback or want me to test out some features with the kids (high schoolers).

abchatra, 2017-11-21:
@sheamusburns Bit late to respond here. Your feedback is invaluable. Thank you so much for writing this. Team is actively debating on the points you made here. I will update you on their status as we go.

abchatra, 2017-11-21:
Here is my first thoughts

`1.	Arrays are a hard concept to grasp with the block syntax. Pop and Push methods in JS are much easier to understand in JS. They really have a tough time conceptualizing an array with the language used on the array blocks. I ended up showing a bunch of them JS arrays and they got it better.`

Push & Pop needs to be the top API’ s we should expose. We do have these API’s but their naming may not be right. I will look into this. 
[image]

 
``1a. Create Empty Array. I sometimes have kids work in blocks, then go take a look at the TS. When they swap back and forth, sometimes something weird happens with their arrays, like all of the suddenly there will be a new block for an array they created: create empty array. Sometimes if the kids swap back and forth between the block and TS modes, repeat create empty array blocks will be created for the same array. Kinda bizarre. If the create empty array block is necessary for the interpreter and the TS, maybe make that block available?``

This sounds like a bug Please file an issue in this repo. We will be happy to look into this.Agreed we need some consistency here. 

``3.	I've noticed in the beta and the stable version that sometimes code just disappears. While it's supposed to save automatically to local storage, there are weird moments where suddenly all the blocks are e...``
This should be fixed by now.

``5.	NeoPixels. Support for the Neopixel library is awesome. It would be great if we could plug and play in the WS2801 Library too!``
Noted.

``6.	I love the new if/then block format with the minus sign and the plus sign. So much better than the little gear. Only thing, I'd improve is when you pull an if true then block out without an else, then add an else, you can't remove the else and turn it back into just an if true then block. That would be nice.``

Bit of limitation on how the blocks are built.

``7.	Functions are awesome, yet they would be infinitely better with params :) Right now I can only teach kids that functions are chunks of code that can be reused. It would be cool to be able to pass parameters and also for a function to return something.``

Noted.