microsoft/pxt-adafruit #493: Editor hangs when you leave any dialog open in the background.

abchatra, 2017-11-08:
Leave in this state for 5 minutes by visiting a different tab
[image]

Then click on discard blocks. Editor hangs. 

[image]

abchatra, 2017-11-08:
I should note I am not reliably able to repro.

abchatra, 2017-11-08:
Happened in edge as well

abchatra, 2017-11-08:
@pelikhan - FYI

microsoftsam, 2017-11-08:
Interesting, I'll keep an eye out for a repro.

microsoftsam, 2017-11-08:
Note, the modal you see there is the semantic JS modal, not the react one. Could be useful info.

microsoftsam, 2017-11-08:
I don't think this has anything to do with the modal. It's more likely it has something to do with console logging. 
When you leave it for 5 minutes, you're writing a bunch of stuff to your buffer. I would start the investigation there.

microsoftsam, 2017-11-08:
@caitlin-hennessy FYI

microsoftsam, 2017-11-08:
In other words, I think this would repro if you leave the editor for 5+ minutes regardless on what screen you're on. Just a hunch

abchatra, 2017-11-08:
It is not console.log related.  There was a error in the program hence the can't decompile dialog

abchatra, 2017-11-08:
I am not able to repro now. I will keep an eye.

microsoftsam, 2017-11-08:
I’ll keep my eyes peeled

pelikhan, 2017-11-08:
[image]

caitlin-hennessy, 2017-11-08:
I will take a look too.

abchatra, 2017-11-09:
Can't repro. Closing

closed 2017-11-09 by abchatra