microsoft/pxt-adafruit #496: blocks background in tutorial should have a background

pelikhan, 2017-11-08:
The block workspace background is stransparent.
[image]

microsoftsam, 2017-11-08:
That was deliberate to keep the "chalkboard" feeling. What background would you give it?

pelikhan, 2017-11-08:
Same color as editor since we know contrast is good there.

microsoftsam, 2017-11-08:
ok

pelikhan, 2017-11-08:
Unrelated but the howl is not really visible in tutorial tab
[image]

pelikhan, 2017-11-08:
black and black, it's cute though

microsoftsam, 2017-11-08:
Yeah maybe we adjust and get rid of it's body, and just keep the eyes :)

pelikhan, 2017-11-08:
Maybe white border around him

microsoftsam, 2017-11-08:
or her?

pelikhan, 2017-11-08:
it

pelikhan, 2017-11-08:
what's her name?

microsoftsam, 2017-11-08:
Minerva

pelikhan, 2017-11-08:
Sounds like a she. It's a she then.

samelhusseini, 2018-04-25:
Closing as contrast only important for text and block color (which is already the case in this design). Also in the docs we don't render a background, why here? 

Reopen if you feel strongly about this.

closed 2018-04-25 by samelhusseini