microsoft/pxt-adafruit #520: IE doesn't have "Animation Editor" under "Light".

Emma-sunny, 2017-11-14:
**Repro Step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Create new project
- Click "_Light_" and choose "_Animation Editor_"

**Expected Result:**
1. IE should have the "Animation Editor" like other browsers.
[image]
2. After open the "_Animation Editor_", The link "Paint your NeoPixel animations" should work.[**Firefox & Chrome**]
3. After click Link "_Open this in a new window_", it should open a page in new window.[**Edge**]

**Actual Result:**
IE:
[image]

Edge:
[image]

FireFox & Chrome:
[image]

pelikhan, 2017-11-14:
Animation Editor is injected by pxt-neoanim.

microsoftsam, 2017-12-29:
As Peli mentioned you would need to add the pxt-neoanim package.

closed 2017-12-29 by microsoftsam