microsoft/pxt-adafruit #521: The inputBox was black when the inputBox was full of comment on Chrome

yanlxu, 2017-11-14:
Repro steps:
• Navigate to https://makecode.adafruit.com/beta
• Drag block "show animation" to "forever" and right click the block to "add comment"
• Input the comment in the inputBox of "comment"

Expected result:
1.The scroll bar can be dragged
2.The inputBox can be displayed

Actual result:
1.The scroll bar can not be dragged on **safari**
2.The inputBox was black when the inputBox was full of comment on **Chrome**

[image]

abchatra, 2017-11-14:
We at overhauling the comment system. This bug needs to wait.

yanlxu, 2018-03-29:
close as p3 priority

closed 2018-03-29 by yanlxu