microsoft/pxt-adafruit #540: Windows MakeCode for CPX App crashes on serial data from the device

abchatra, 2017-11-21:
Attach the device and flash the following program
[image]

Play with the device for a while. Data is getting streamed from the device. There will be a native null pointer exception from the app after say 10 min. You might have to minimize the app and reopen.

Here is the heap dump of the process
`\\scratch2\scratch\abchatra\MemoryDump`

Here is the stack of the crash:
```C++
:003> k20
  Child-SP          RetAddr           Call Site
00 000000a1`ae3fdfd0 00007ffb`a74e638c Windows_Devices_HumanInterfaceDevice!HidUtil::InitializeReport+0x37 [onecoreuap\drivers\input\hid\client\winrt\lib\hidutils.cpp @ 50] 
01 000000a1`ae3fe010 00007ffb`8ae3924e Windows_Devices_HumanInterfaceDevice!Windows::Devices::HumanInterfaceDevice::HidDeviceServer::CreateOutputReportById+0x3c [onecoreuap\drivers\input\hid\client\winrt\lib\hiddevice.cpp @ 1176] 
02 000000a1`ae3fe070 00007ffb`8ac873bb Chakra!amd64_ProjectionCall+0x7e [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\amd64\projectioncall.asm @ 127] 
03 000000a1`ae3fe0d0 00007ffb`8ac86f30 Chakra!Projection::ProjectionMethodInvoker::InvokeUnknown+0xcb [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\projectionmethodinvoker.cpp @ 386] 
04 000000a1`ae3fe160 00007ffb`8ae26aee Chakra!Projection::InvokeMethodByThisInfo+0x170 [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\projectionmarshaler.cpp @ 384] 
05 000000a1`ae3fe2e0 00007ffb`8ae26a70 Chakra!Projection::DynamicCall+0x76 [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\projectionmarshaler.cpp @ 447] 
06 000000a1`ae3fe330 00007ffb`8ae36df6 Chakra!Projection::MethodSignatureThunk+0x40 [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\projectionmarshaler.cpp @ 466] 
07 000000a1`ae3fe370 00007ffb`8ad0942c Chakra!amd64_CallFunction+0x86 [d:\rs2\onecoreuap\inetcore\jscript\core\lib\Runtime\Library\amd64\JavascriptFunctionA.asm @ 208] 
08 000000a1`ae3fe3c0 00007ffb`8ad08e46 Chakra!Js::JavascriptFunction::CallFunction<1>+0x9c [d:\rs2\onecoreuap\inetcore\jscript\core\lib\runtime\library\javascriptfunction.cpp @ 1366] 
09 000000a1`ae3fe420 00007ffb`8ae2c41f Chakra!Js::JavascriptFunction::CallFunction+0x36 [d:\rs2\onecoreuap\inetcore\jscript\core\lib\runtime\library\javascriptfunction.cpp @ 1113] 
0a 000000a1`ae3fe460 0000028a`0c4f15e7 Chakra!Projection::OverloadMethodThunk+0xaf [d:\rs2\onecoreuap\inetcore\jscript\private\lib\projection\projectionwriter.cpp @ 201] 

0:003> ?? preparsedData
struct _HIDP_PREPARSED_DATA * 0x00000000`00000000

0:003> .load \\scratch2\scratch\abchatra\MemoryDump\jd.dll

0:003> !jd.jstack
0b js!WindowsRuntimeIO.prototype.sendPacketAsync (#11.30, #18631) [JIT]
0c js!loop (#9.562, #24001) [JIT]
0d js!Anonymous function (#9.564, #24002) [JIT]
0e js!Anonymous function (#9.109, #24003) [JIT]
0f js!Promise.prototype._execute (#15.2216, #20915) [JIT]
10 js!Promise.prototype._resolveFromExecutor (#15.2428, #20738) [JIT]
11 js!Wrapper.prototype.talkAsync (#9.555, #17976) [JIT]
12 js!Anonymous function (#15.303, #24015) [JIT]
```

abchatra, 2018-06-06:
@guillaumejenkins I am assuming this is fixed. Right?

closed 2018-06-06 by abchatra

guillaumejenkins, 2018-06-07:
Yes this was the bug where the app would be suspended from being minimized, and we then tried to reuse the HID handles (which had been invalidated). We fixed this by releasing all connections when the app gets suspended.