microsoft/pxt-adafruit #558: Console Access Timing Bug & Printing Numbers As Strings Not Possible

sheamusburns, 2017-12-06:
Heya, 

We've been running into a problem: there's no way to access the console before an event fires which calls the console object to make the console simulator button appear. By the time that button appears, the event that we're trying to monitor may have already happened though. Need a way to enter into the console whenever the user wants. 
One way we've hacked around this is to call console.log('initialize console') on start, which will give us a back-door into the console so we can monitor the target event. 

Because this is easier to see in a workflow, I've created some sample code: [https://makecode.com/_K5UimkKmiPa3](https://makecode.com/_K5UimkKmiPa3)
...and made a short video walking through the use case around the issue:
[http://www.youtube.com/watch?v=cnpadiAc7q0](http://www.youtube.com/watch?v=cnpadiAc7q0)

Side note -- Can you add a `toString()` method block? Sometimes just want to print out a handful of number values to the console without taking up all the real estate that `console log value` takes up. Since we can't print out numbers in the `console log` block, this makes it nearly impossible. It could be fixed by having a `Number To String` block. 

Thanks!

pelikhan, 2017-12-06:
This is by design currently but we can probably fix it. Meanwhile try hitting the restart button under the simualtor.

http://www.youtube.com/watch?v=cnpadiAc7q0<https://na01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DcnpadiAc7q0&data=02%7C01%7Cjhalleux%40microsoft.com%7C5df7679e73a24631fe3d08d53c501ebf%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636481237169795953&sdata=NUoaJ4DrQwuVKGXphY3cH2TzK7WvTUAGBTH2mZ7kkDU%3D&reserved=0>

sheamusburns, 2017-12-06:
Thanks. Will do. 

S

pelikhan, 2017-12-06:
Can you open another bug for the sidenote? :)

pelikhan, 2017-12-20:
This PR should fix this. https://github.com/Microsoft/pxt/pull/3656/files