microsoft/pxt-adafruit #573: much better mic audio

ladyada, 2017-12-13:
hiya - i totally jammed out last weekend and got PDM microphone support working much better, with a faster filter and the use of DMA. i removed all ASF bindings so its completely 'free standing' 

the audio is _a bit_ staticky but works very well. we sample at 16khz, but you can get reasonable audio at 8khz - im working on a demo that will write the audio to the SPI flash but im not ready quite yet :)

https://github.com/adafruit/Adafruit_ZeroPDM

pelikhan, 2017-12-13:
@mmoskal fyi

mmoskal, 2017-12-13:
Looks cool! I think something to be integrated into codal @jamesadevine @finneyj

As for SPI flash I'm working on a file system support with wear-leveling for the CPX so when we get mic support we'll just use that.
https://github.com/lancaster-university/codal-core/blob/snorfs/source/drivers/SNORFS.cpp

ladyada, 2017-12-13:
neat! the [SPI flash we uses is guaranteed for 100K cycles](http://www.gigadevice.com/product/download/365.html?locale=en_US) so you don't have to worry *a ton* about wear leveling ([experiments have shown you can get 10M cycles](http://dangerousprototypes.com/blog/2010/06/10/flash-destroyer-wrap-up/) ). 

would you be OK with using the circuitpython-compatible file system? if so people can use an UF2 to get/retreive files. [see here](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/using-spi-flash#accessing-spi-flash)

mmoskal, 2017-12-13:
We want to use it for data logging. If you log a value every 5s, you will get to 100k erases of the FAT after 1 week (without even filling up the flash). I understand that you might be able to get more than 100k but it's not guaranteed and depends on things like voltage and ambient temperature. Also, erasing a sector is kind of slow, and you need to do that quite a bit when using FAT.

ladyada, 2017-12-13:
hmm i'd like to try to maintain some compatibility with the FS that ships on the board, if possible! makes USB easier as well....

what if we keep the FAT FS BUT you get/create one big file, say "MakeCode.bin" or .dat or whatnot, that is 512KB fixed size. then you write all you want to the file, and you wont have to wearlevel it because it isn't being erased 

would that address the concern?

pelikhan, 2017-12-13:
Eucators/students will expect to connect the CPX and see their “data.csv” file in there, double click and analyze the data in Excel. That flow enlables essay STEM activities. I don’t think cross compat is really needed for that?

tannewt, 2017-12-13:
@pelikhan is there a better place to discuss the FS stuff? I'm curious to know more details but this issue should be related to mic stuff.

pelikhan, 2017-12-13:
For sure, @mmoskal do you have a PR/issue we can latch on?

mmoskal, 2017-12-13:
Let's use this: https://github.com/lancaster-university/codal-core/pull/6

ladyada, 2018-01-22:
adding a note that for arduino side of CPX we implemented a 'soundpressurelevel' function that is useful for having things change when audio volume changes

https://github.com/adafruit/Adafruit_CircuitPlayground/blob/master/utility/Adafruit_CPlay_Mic.cpp#L160

pelikhan, 2018-04-03:
@ladyada is it possible to migrate the sound pressure level to CODAL?

ladyada, 2018-04-03:
https://github.com/lancaster-university/codal-samd21/pull/3

mmoskal, 2018-04-03:
Now merged.

abchatra, 2018-06-07:
@mmoskal is this issue completed?

mmoskal, 2018-06-07:
Need to tag codal and bump in pxt

pelikhan, 2018-06-15:
New sound level integrated.

closed 2018-06-15 by pelikhan