microsoft/pxt-adafruit #574: USB keyboard

ladyada, 2017-12-13:
hiya @Jaqster @pelikhan  we'd like to add USB keyboard (HID) support. this can be 'hacked' to do datalogging, MIDI, and of course BANANA PIANOs :)

we see https://github.com/lancaster-university/codal-samd21/blob/master/source/USB.cpp some action going on here.

two Qs:
1. whats the best way to do that kind of low level development?
2. is there a place where we can find the rest of the USB code?

Jaqster, 2017-12-13:
Awesome!!  That would be great :-)  Yeah, we can set up a quick call with Joe to go through the CODAL repo and explain how you could add an extension.  Will look for some time.

pelikhan, 2017-12-13:
1) Adafruit could publish new driver/low level code compatible with CODAL in github repos. We would then be able to link them into MakeCode in a sustainable way. Would love to talk about enabling this for you.
2) This is where we use the CodalUSB device in MakeCode: https://github.com/Microsoft/pxt-common-packages/blob/c0d078f79b46c3349453c45c945c1ec1e22d6128/libs/core/usb.cpp

pelikhan, 2017-12-13:
@mmoskal

mmoskal, 2017-12-13:
@ladyada USB.cpp is just the SAMD21 specific driver. There is a generic HID interface here https://github.com/lancaster-university/codal-core/blob/master/inc/drivers/HID.h and https://github.com/lancaster-university/codal-core/blob/master/source/drivers/HID.cpp

We have a custom HID interface here https://github.com/Microsoft/pxt-common-packages/blob/master/libs/core/hf2.cpp and https://github.com/Microsoft/pxt-common-packages/blob/master/libs/core/hf2.h

It should be relatively easy to add a HID Keyboard, probably in Codal, not in pxt-common-packages.

ladyada, 2017-12-13:
kk - we have to change the custom descriptor and reports. would you rather we piggyback on the current HID interface or have a composite device?

mmoskal, 2017-12-13:
Adding another `CodalUSBInterface` (including one derived from `USBHID `) will automatically create a composite device, so no need to piggyback.

mmoskal, 2017-12-13:
i.e., we can have the current `HF2` interface, and a new Keyboard one derived from USBHID (or `CodalUSBInterface` for that matter), and by adding both of them to CodalUSB we will have a composite device.

ladyada, 2017-12-13:
awesome. i didnt know it was pluggable USB. thats awesomepants :) should we have the keyboard interface active no matter what or would the existance of an 'onstart' block cause it to be added (e.g. in python we must init USB before the usercode starts so all pyboards have USB HID/MSD/CDC composites)

ladyada, 2017-12-13:
awesome. i didnt know it was pluggable USB. thats awesomepants :) should we have the keyboard interface active no matter what or would the existance of an 'onstart' block cause it to be added (e.g. in python we must init USB before the usercode starts so all pyboards have USB HID/MSD/CDC composites) ?

mmoskal, 2017-12-13:
This would be decided on PXT not Codal side. Codal doesn't support dynamic adding of interfaces, but I think neither does USB protocol in general (short of doing a full USB reset). In PXT we could delay USB initialization for say 20ms and then decide what interfaces to include. I think that would be nicer, because at some point we'll start to run out of USB endpoints.

ladyada, 2017-12-13:
OK!

ladyada, 2017-12-21:
hiya, good news! @deanm1278 has added USB HID keyboard support

https://github.com/adafruit/codal-core/commit/bfe5145d949646e33ca94b463733bffd71b25ebc

and has a basic demo

```
#include "CircuitPlayground.h"
CircuitPlayground cplay;
#include "HIDKeyboard.h"
CodalUSB usb;
USBHIDKeyboard hid;
static const char *string_descriptors[] = {
    "Example Corp.", "PXT Device", "42424242",
};
int main()
{
    int status = 0;
    usb.stringDescriptors = string_descriptors;
    usb.add(hid);
    usb.start();
    uint8_t key = 0;
    //we need a bit of a delay for USB
    cplay.sleep(500);
    while(1)
    {
        cplay.io.led.setDigitalValue(status);
        //press a key
        hid.keyPress(key);
        cplay.sleep(100);
        //release the key
        hid.keyUp();
        key++;
        cplay.sleep(800);
        status = !status;
    }
}
```

our next question is - what is the API you want to this from typescript. e.g. should we just have a keycode-sender and typescript does the character-to-keycode conversion? OR we can take in an ascii string and do it all for ya. just depends where you want to split the labor. 

note that each country has a different keymapping >.< we could do US for sure but other countries would be a little challenging for us to properly test

pelikhan, 2017-12-21:
Something that maps to https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/key/Key_Values would be great

adafruit/codal-core@bfe5145<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fadafruit%2Fcodal-core%2Fcommit%2Fbfe5145d949646e33ca94b463733bffd71b25ebc&data=02%7C01%7Cjhalleux%40microsoft.com%7C7ea948713ca54788e16008d5489559fa%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636494728667347260&sdata=iCCNmVQEgl5LcjCqua3e%2BuxU77HSNeEsko74boqJjVg%3D&reserved=0>

mmoskal, 2017-12-21:
IMHO the more we put in TypeScript the better especially for more esoteric features. The C++ code generally gets linked every time whereas Typescript is only linked as needed.

deanm1278, 2017-12-21:
> KeyboardEvent/key/Key_Values would be great
> com/?url=https%3A%2F%2Fgithub.com%2Fdeanm1278&data=02%7C01%
> 7Cjhalleux%40microsoft.com%7C7ea948713ca54788e16008d5489559fa%
> 7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636494728667347260&sdata=Pk%
> 2BZ1o0xDz26eJTdGDFsSq1cAvMLx0ErUPZZrYT5cYU%3D&reserved=0> has added USB
> protection.outlook.com/?url=https%3A%2F%2Fgithub.com%
> 2Fadafruit%2Fcodal-core%2Fcommit%2Fbfe5145d949646e33ca94b463733
> bffd71b25ebc&data=02%7C01%7Cjhalleux%40microsoft.com%
> 7C7ea948713ca54788e16008d5489559fa%7C72f988bf86f141af91ab2d7cd011
> db47%7C1%7C0%7C636494728667347260&sdata=iCCNmVQEgl5LcjCqua3e%
> 2FMicrosoft%2Fpxt-adafruit%2Fissues%2F574%23issuecomment-
> 353404463&data=02%7C01%7Cjhalleux%40microsoft.com%
> db47%7C1%7C0%7C636494728667347260&sdata=s%2B7GdDF66w5ANkeyHzrGn3BbhmyB7t
> 5ZRE0BZMLu0gH1ks5tCpBegaJpZM4RA6Mv&data=02%7C01%7Cjhalleux%
> 40microsoft.com%7C7ea948713ca54788e16008d5489559fa%
> 7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636494728667347260&sdata=

What if the C++ API was just
>
>  keyDown(uint8_t key, uint8_t report=GENERIC);

 keyDown(uint8_t key, uint8_t report=GENERIC);

Where GENERIC would be regular key codes and CONSUMER would be multimedia
keys

C++ would manage the internal state of which keys are pressed, and you
would basically do in typescript what you do with your fingers on the
keyboard:
pressKey('shift'), pressKey('a'), releaseKey('a'), releaseKey('shift')

deanm1278, 2017-12-21:
> en-US/docs/Web/API/KeyboardEvent/key/Key_Values would be great
> nks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%

sorry typo, C++ API would be:
>
>  keyDown(uint8_t key, uint8_t report=GENERIC);

 keyUp(uint8_t key, uint8_t report=GENERIC);

deanm1278, 2017-12-21:
also just to clarify intent, typescript would be doing all the keycode
lookups. C++ would just feed through the passed keycodes to the host
machine.

deanm1278, 2017-12-22:
an implemented API is here for your consideration:
https://github.com/adafruit/codal-core/commit/17cf9b41c94c81ec697b9ad249268f9aa5567db2

        int modifierKeyDown(uint8_t key, uint8_t
> reportID=HID_KEYBOARD_REPORT_GENERIC);
>         int modifierKeyUp(uint8_t key, uint8_t
> reportID=HID_KEYBOARD_REPORT_GENERIC);
>         int keyDown(uint8_t key, uint8_t
> reportID=HID_KEYBOARD_REPORT_GENERIC);
>         int keyUp(uint8_t key, uint8_t
> reportID=HID_KEYBOARD_REPORT_GENERIC);

Typescript would decide whether the key sent was a modifer (e.x.
pressKey('LSHIFT') ) or a normal key (e.x. pressKey('A') )

The main.cpp example is:

#include "CircuitPlayground.h"

> CircuitPlayground cplay;

> #include "HIDKeyboard.h"

CodalUSB usb;

USBHIDKeyboard hid;

> static const char *string_descriptors[] = {

    "Example Corp.", "PXT Device", "42424242",

};

> int main()

{

    usb.stringDescriptors = string_descriptors;

    usb.add(hid);

    usb.start();

>     uint8_t key = 0;

>     //we need a bit of a delay for USB

    cplay.sleep(500);

>     while(1)

    {

        //hold left shift key

        hid.modifierKeyDown(0x02);

>         //press a key

        hid.keyDown(key);

>         cplay.sleep(100);

        //release the key

        hid.keyUp(key);

        key++;

>         //release left shift key

        hid.modifierKeyUp(0x02);

>         cplay.sleep(800);

    }

}

deanm1278, 2017-12-27:
> com/adafruit/codal-core/commit/17cf9b41c94c81ec697b9ad249268f9aa5567db2
> fications/unsubscribe-auth/AKLfOGGSi2Bfm8hdk_N52qYK9nCWcUZgk

Hello! just pinging this thread again, I have added HID mouse and HID
gamepad as well:
https://github.com/adafruit/codal-core

The mouse implementation has 3 buttons, wheel, and xy movement and the
interface is:

int buttonDown(USBHIDMouseButton b);
> int buttonUp(USBHIDMouseButton b);
> int move(int8_t x, int8_t y);
> int moveWheel(int8_t w);

The gamepad has 2 joysticks, 2 throttles (values 0-31) and 16 buttons. This
layout can of course be changed. The interface is:

int buttonDown(uint8_t b);
> int buttonUp(uint8_t b);
> int move(int8_t num, int8_t x, int8_t y);
> int setThrottle(uint8_t num, uint8_t val);

pelikhan, 2017-12-27:
Thanks! Most of the teams are on holiday break so this will have to wait till Jan.

https://github.com/adafruit/codal-core

deanm1278, 2017-12-27:
> 354166207&data=02%7C01%7Cjhalleux%40microsoft.com%
> 7C953513993d49493bb99908d54d60c49a%7C72f988bf86f141af91ab2d7cd011
> db47%7C1%7C0%7C636500000372779927&sdata=VahpOaMdZBBBNthDN4ZMLl76swGOL7
> s5tEpuhgaJpZM4RA6Mv&data=02%7C01%7Cjhalleux%40microsoft.com%
> db47%7C1%7C0%7C636500000372779927&sdata=2aGf0d4Wtn6y1Znidn90pO14RXw7o%

no worries! We'll hold off on any more updates until then.

ladyada, 2018-01-02:
@pelikhan wooo happy new year! lets do this thing - we have this exciting code written and ready to merge in and implement, are you ready? :)

pelikhan, 2018-01-02:
@deanm1278 @ladyada the C++ api looks great. The next step is to get it PR/Reviewed/merged in codal-core. I'll start building the TypeScript layer around it.

ladyada, 2018-01-02:
ok should dean submit it as a PR? or is that something you do?

pelikhan, 2018-01-02:
I think you should submit the PR for CODAL-core.

ladyada, 2018-01-02:
great - its submitted :)

pelikhan, 2018-01-02:
@ladyada there is an alternative route to sending a PR to CODAL-core which is creating your own repos:

* create a new repo under your github org that exposes the USB HID C++ code (eg. adafruit/codal-usbhid).
* create a new repo that contains the makecode gluing code (e.g. adafruit/pxt-usbhid). Once setup, the user can add this package using the "Extensions" button from the editor and get the new blocks.

Benefits: there is no bottleneck to deliver new drivers (like people taking the holidays off :)), new blocks can be delivered through the "Extensions" feature and tested. The most popular ones can be preloaded eventually

@finneyj

ladyada, 2018-01-02:
i thought it would be kinda cool to have it be built-in by default, rather than an extension? whichever way y'all prefer!

pelikhan, 2018-01-02:
Yeah, USBHID is a special universal one... Other driver might fall under the "do your own repo" kind of category, similarly to your various Arduino drivers.

ladyada, 2018-01-02:
ok going forward for specialized drivers we will do so! :)

pelikhan, 2018-01-03:
Happy to help with the plumbing for the next one :)

mmoskal, 2018-01-03:
To clarify: I don't think we support a separate C++ repo per extension, however a `pxt-my-driver` repo can contain C++ code.

closed 2018-04-03 by pelikhan